SpikeInterface/spikeextractors #593: 'CommonReferenceRecording' object has no attribute 'has_unscaled'

luiztauffer, 2021-03-05:
Example:
```
recording_processed = st.preprocessing.bandpass_filter(recording, freq_min=freq_min_hp, freq_max=freq_max_hp)
recording_processed = st.preprocessing.common_reference(recording_processed)
recording_processed = se.SubRecordingExtractor(recording_processed, end_frame=150000)
```

Error:
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-13-509cdf7ab5c6> in <module>
      1 recording_processed = st.preprocessing.bandpass_filter(recording, freq_min=freq_min_hp, freq_max=freq_max_hp)
      2 recording_processed = st.preprocessing.common_reference(recording_processed)
----> 3 recording_processed = se.SubRecordingExtractor(recording_processed, end_frame=150000)
      4 
      5 

~/anaconda3/envs/env_klusta/lib/python3.7/site-packages/spikeextractors/subrecordingextractor.py in __init__(self, parent_recording, channel_ids, renamed_channel_ids, start_frame, end_frame)
     14         self._start_frame = start_frame
     15         self._end_frame = end_frame
---> 16         self.has_unscaled = self._parent_recording.has_unscaled
     17         if self._channel_ids is None:
     18             self._channel_ids = self._parent_recording.get_channel_ids()

AttributeError: 'CommonReferenceRecording' object has no attribute 'has_unscaled'
```

@alejoe91 I am not sure if has_unscaled should or should not be there, so I'm not sure if this Issue belongs to Spikeextractors or Spiketoolkit

alejoe91, 2021-03-05:
@luiztauffer you need to use this branch of spiketoolkit https://github.com/SpikeInterface/spiketoolkit/pull/440

luiztauffer, 2021-03-05:
I am using that branch...

luiztauffer, 2021-03-05:
alright, I'm sorry about that, for some reason I've installed the pip version on my env... problem solved

closed 2021-03-05 by luiztauffer