SpikeInterface/spikeextractors #617: error opening blackrock .nev files

Saksham20, 2021-04-06:
Opening a set of Blackrock files (.nev, .ns6) and then calling the method `get_traces()` throws an error: 
```
Traceback (most recent call last):
  File "C:\Users\Saksham\Documents\NWB\roiextractors\venv\lib\site-packages\IPython\core\interactiveshell.py", line 3418, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-5-b0b7e6504b87>", line 1, in <module>
    bk.get_traces()
  File "c:\users\saksham\documents\nwb\spikeextractors\spikeextractors\extraction_tools.py", line 796, in corrected_args
    traces = func(recording, channel_ids=channel_ids, start_frame=start_frame, end_frame=end_frame,
  File "c:\users\saksham\documents\nwb\spikeextractors\spikeextractors\extractors\neoextractors\neobaseextractor.py", line 106, in get_traces
    raw_traces = self.neo_reader.get_analogsignal_chunk(block_index=self.block_index, seg_index=self.seg_index,
TypeError: get_analogsignal_chunk() got an unexpected keyword argument 'stream_index'
```
These are the .nev/ns6 files for reference: 
[under sample data](https://www.blackrockmicro.com/technical-support/software-downloads/)

alejoe91, 2021-04-06:
@Saksham20 what version of neo are you running? Can you try with neo master? Anyways it shouldn't fail, as we have different behaviors for different versions. But probably something is off!

samuelgarcia, 2021-04-06:
I think you need to update neo to the master branch.

alejoe91, 2021-04-06:
@samuelgarcia still it shouldnt fail.

@Saksham20 do you have SE master?

Saksham20, 2021-04-06:
> @samuelgarcia still it shouldnt fail.
> 
> @Saksham20 do you have SE master?

yes, its master. The method `get_analogsignal_chunk()` in master does not have `block_index `as an argument

bendichter, 2021-04-14:
this has been resolved. Please close

closed 2021-05-01 by Saksham20