SpikeInterface/spikeextractors #638: module 'spikeinterface.extractors' has no attribute 'NumpyRecording'

nzgurel, 2021-05-13:
Hello, I'm new to spikeinterface, all packages are installed. I am going through this tutorial:
https://spikeinterface.readthedocs.io/en/latest/modules/core/plot_1_recording_extractor.html#sphx-glr-modules-core-plot-1-recording-extractor-py

I have an error when trying to use the NumpyRecording attribute in extractors:

```
recording = se.NumpyRecording(traces_list=[traces0, traces1], sampling_frequency=sampling_frequency)
print(recording)
```
module 'spikeinterface.extractors' has no attribute 'NumpyRecording'

samuelgarcia, 2021-05-17:
Please not that spikeinterface is ina process of a big refactoring:
  * next version  0.90.0
    * API break
    * doc is here https://spikeinterface.readthedocs.io/en/latest
    * install from source only
    * officla release this summer

  * acutal release 0.12.0
    * legacy API
    * doc is here https://spikeinterface.readthedocs.io/en/stable/
    * install with `pip install spikeinterface`

If you go for the new API (good choice), you can except some bugs and you need will need to re install from timt to time.

nzgurel, 2021-05-23:
Thank you so much for the detailed info, I'll try the new API.

closed 2021-05-23 by nzgurel