SpikeInterface/spikeextractors #647: Comparison SE/neo/mat73 on spike2 files

eduardacenteno, 2021-06-15:
Hey guys, 
I'm trying to work with >2GB files in Spike2, thus I'm trying to move from .smr to .smrx. 
I tried using the CED extractor here in SE, and made some comparisons with neo's CED IO recently implemented by @samuelgarcia , and a .mat version of the same file which I read with the package [mat73](https://pypi.org/project/mat73/).

My problem is, when I get the same array from these 3 different sources, the output is different and might or might not work in the rest of the analysis I have to do in the lab. (If I plot with matplotlib, the signals look the same.. except the y  range ofc)

Output: 
- from mat file using mat73: 'values': array([-1.95318462e-05, -1.35807368e-05, -1.27872554e-05, ...,
         1.35502187e-05, -1.31229592e-06,  4.66933216e-06])}

- from smrx file using SE: array([[-640., -445., -419., ...,  623.,  444.,  -43.]], dtype=float32)

- from neo: array([[-0.12800391],
       [-0.08900272],
       [-0.08380256],
       ...,
       [ 0.08880271],
       [-0.00860026],
       [ 0.03060094]], dtype=float32)

The only one that works for my further analysis (spectral analysis using [HVC](https://hybrid-vocal-classifier.readthedocs.io/en/latest/index.html)) is the first (from the mat file ... or if I export the song channel to .smr and work with neo -- that works as well)

Any comments on that?
I'm trying to write a script to get smrx data in the lab, but now I'm not sure if I should consider exporting them to .mat instead..

[image]

samuelgarcia, 2021-06-15:
* for matlab source, are you sure that the matlab reader handle correctly units ?
 * For spikeextractor did you used `return_scaled=True` ? It seams like the values are not scaled to micro volt in that case but kept as int16
 * for neo.io method, you should able to get units for analogsignals.

You should open the file with spike2 software if you have acces to it and check which one is correct about units and magnitude.
If the neo one, please tell us quiclky!!

eduardacenteno, 2021-06-15:
Hey @samuelgarcia !

Yes, I used return_scaled=True in SE. 

From what I see in Spike2, the matfile is the most accurate.
[image]

alejoe91, 2021-06-15:
Thanks @eduardacenteno 

I'll add @luiztauffer to the discussion too

eduardacenteno, 2021-06-16:
Hey, just had a chat with @samuelgarcia and we fixed it for neo. Basically the gain had to be divided by a number given in sonpy documentation. gain = smrx.GetChannelScale(chan_ind) / 6553.6
Now it matches the .mat and what we see in spike2.

For SE, I'd say the scaling also needs to be done accordingly:

user units = integer * scale / 6553.6 + offset

'The factor of 6553.6 comes about because a scale value of 1.0 converts between an input range of 10 Volts and an
ADC range of 65536. In most cases, you should not need to worry about this, just remember to use the equations.' Source: SONPY documentation

luiztauffer, 2021-06-23:
@eduardacenteno thanks for finding this!
#651 should fix it for SE

PR #651 (2021-06-28): CED set gains
Fix #647 

- Set gain values as `gain` instead of `ced_gain`
- Set `self.has_unscaled = True` for decorator `@check_get_traces_args` to work
closed 2021-06-28 by alejoe91